adafruit/Fritzing-Library #41: Parts in the Schematic view are too big.

htho, 2015-01-13:
Hi,

in the schematic view, many of the probably "older" parts in the library are too big in relation to the other parts. I have attached a screenshot. All "older" parts including the "TPA2016 Stereo Amplifier" are about three times as big as they should. The "TMP007" and later parts are in scale.

Maybe there is an easy way to fix it?

Best

htho
[image]

rellimmot, 2015-03-28:
+1. Would like a fix or workaround.
[image]

aknoerig, 2015-03-31:
We changed the default schematic size to a more common 0.1" grid about a year ago. 
See this blog post: http://blog.fritzing.org/2014/01/23/0-8-6-schematics/

Not all libraries have been updated since then. There is a command in the parts editor that helps you with the transition, see the instructions in the post.
Also all our scripts have been updated, and there is also one that does a bulk conversion.

rellimmot, 2015-03-31:
I understand. The "convert schematic to 0.1 grid option" is not available for me. I'm using Fritzing v0.9.1 running under OS X 10.10.1. 

I obtained the library with the problematic parts from here: https://codeload.github.com/adafruit/Fritzing-Library/zip/master

aknoerig, 2015-04-03:
Sorry, the blog post isn't very precise. The command can be found under the parts editor's "Edit" menu, not the file menu.

ladyada, 2016-07-12:
we'll get to this at some point - all *new* devices being made are 0.1" and we are remaking some old ones. closing because we dont guarantee all items are 0.1"

closed 2016-07-12 by ladyada