adafruit/Fritzing-Library #42: ULN2803 Darlington Array is incorrect ...

DougieLawson, 2015-01-15:
If I create a sketch with a ULN2803 the GND and COM pins are switched in the schematic.

XML has 

        <connector id="connector9" type="male" name="COMMON">
            <description>no description yet</description>
            <views>
                <breadboardView>
                    <p layer="breadboard" svgId="connector9pin"/>
                </breadboardView>
                <schematicView>
                    <p layer="schematic" svgId="connector9pin"/>
                </schematicView>
                <pcbView>
                    <p layer="copper0" svgId="connector9pin"/>
                </pcbView>
            </views>
        </connector>

and  
        <connector id="connector10" type="male" name="GND">
            <description>no description yet</description>
            <views>
                <breadboardView>
                    <p layer="breadboard" svgId="connector10pin"/>
                </breadboardView>
                <schematicView>
                    <p layer="schematic" svgId="connector10pin"/>
                </schematicView>
                <pcbView>
                    <p layer="copper0" svgId="connector10pin"/>
                </pcbView>
            </views>
        </connector>

Which are the wrong way round.

DougieLawson, 2015-01-15:
I found the other part of the bug in svg.schematic.ULN2803_Darlington_Array__b60cbac200ab61__schematic__d35296a78c3ae0acf55759a77708a2ee.svg

```xml
    <rect
       width="7.0866141"
       height="7.0866141"
       x="182.48032"
       y="262.20465"
       id="connector10pin"
       style="fill:none;stroke:none" />
    <rect
       width="7.0866141"
       height="7.0866141"
       x="-30.118116"
       y="-269.29138"
       transform="scale(-1,-1)"
       id="connector9pin"
       style="fill:none;stroke:none" />
```
Needs to change to

```xml
    <rect
       width="7.0866141"
       height="7.0866141"
       x="182.48032"
       y="262.20465"
       id="connector9pin"
       style="fill:none;stroke:none" />
    <rect
       width="7.0866141"
       height="7.0866141"
       x="-30.118116"
       y="-269.29138"
       transform="scale(-1,-1)"
       id="connector10pin"
       style="fill:none;stroke:none" />
```

ladyada, 2016-08-27:
fixed!

closed 2016-08-27 by ladyada