adafruit/Fritzing-Library #46: T Cobbler Plus pins are mislabeled/wrong

nkolban, 2015-03-30:
Using the latest Fritzing parts I find that the T-Cobbler Plus pins are mislabeled.  To see this, open up the T-Cobbler Plus and open the parts editor.  Select the pin labeled on the board as "#20" and see that it is actually flagged as GND.  There are many other pins mislabeled.  The board image is correct but the connector list is wrong.

See attached screen shot for an example.

[image]

nkolban, 2015-05-22:
Am I able to make the changes myself to this Github project and check-in the changes?  I don't want to waste time fixing things if I don't have permissions to check changes back in.

matt-cook, 2015-10-06:
@nkolban you'll want to make a fork of the repository into your own account, push the changes to that and then create a pull request: https://help.github.com/articles/using-pull-requests/

ladyada, 2016-08-27:
they've all been redone!

closed 2016-08-27 by ladyada