adafruit/Fritzing-Library #47: Pi cobblers don't produce reliable ratsnest in PCB

patrickfuller, 2015-06-30:
As an example, if you set up this breadboard:

[image]

the PCB ratsnest connects two unrelated pins:

[image]

nkolban, 2015-06-30:
Howdy Patrick, see also issue #46 I get the feeling that there is no-one on duty watching these issues.  You might want to modify the Fritizing parts yourself if you are in a hurry.

patrickfuller, 2015-06-30:
If no one's watching these issues, then that sounds like the best way to go.

I would still like to make sure that the Pi cobbler works for everyone. I'll send an email and see if I have any luck.

patrickfuller, 2015-07-02:
Posted this on [adafruit's forum](https://forums.adafruit.com/viewtopic.php?f=50&t=76186&p=386486#p386486). Hopefully this gets resolved.

nkolban, 2015-07-02:
Good idea.  Wish I'd thought to do that myself with the issue I raised ... I just sat here mumbling that Adafruit weren't watching the issues :-)

ladyada, 2016-08-27:
they've all been redone!

closed 2016-08-27 by ladyada