adafruit/Fritzing-Library #49: Fritzing files request for TSL2591, TSL2561, Metro 328, and Metro 328 Mini

ashfaque-s, 2015-07-06:
Kindly requesting Fritzing files for the following Adafruit items:
(1) TSL2561 (ambient light sensors): http://www.adafruit.com/products/439 
(2) TSL2591 (ALS): https://www.adafruit.com/products/1980
(3) Metro 328: https://www.adafruit.com/product/2466
(4) Metro Mini 328: https://www.adafruit.com/product/2590

Thanks :)

nkolban, 2015-07-06:
Adafruit create their own Fritzing libraries for their own products.  I'd be delighted to build these for you but that would just be wasted time as Adafruit would then release their own and we would have duplicates.   What I suggest you do is post an issue to the Adafruit Fritzing Github to request that they add these parts to their own Fritizing parts libraries.

See:

https://github.com/adafruit/Fritzing-Library

alexissantos, 2016-03-01:
+1 to this. Would love a file for the Metro Mini.

Wifsimster, 2016-04-08:
+1 for the TSL2561.

a432511, 2016-04-18:
+1 Metro Mini please!

ladyada, 2016-07-04:
Metro mini: https://github.com/adafruit/Fritzing-Library/commit/a82a6beae416e95174d685e2c48576e3b3bc1374

ladyada, 2016-07-04:
TSL's https://github.com/adafruit/Fritzing-Library/commit/e307aeb3e4cd97a1db7f3afe87898916367f781e

closed 2016-07-04 by ladyada