adafruit/Fritzing-Library #81: Schematic images are too big

deshipu, 2016-08-11:
I'm preparing a workshop in which, among other things, I want to teach people to read proper schematics. The idea is to show both a breadboard view and schematic view of the same circuits. However, all the adafruit parts have the schematic image much larger than reasonable, meaning all the other parts become too small to comfortably recognize.

ladyada, 2016-08-12:
see https://github.com/adafruit/Fritzing-Library/issues/41
we are re-doing the older devices but theres *a lot* - if you have a specific request, open a ticket for those parts
all new parts follow the 2014+ standard

closed 2016-08-12 by ladyada

deshipu, 2016-08-12:
Thank you for pointing me to that issue, it has a pointer to how to solve it.