adafruit/Fritzing-Library #99: RGB LCD Displays Fail on Grid Conversion

andrewvaughan, 2017-06-14:
The Adafruit RGB 2x16 LCD Positive (and likely Negative) displays do not follow the 0.1 grid standard. As instructed in #81, I am opening a ticket to have that part be resized.

Furthermore, when I attempt to resize the part myself, Fritzing spits out the following errors:

On Edit:
> This version of the new Parts Editor can not deal with separate copper0 and copper 1 layers in '/.../Adafruit_Positive_RGB_2x16_LCD_....svg'. So editing may produce an invalid PCB view image.

On Conversion:
> Missing connector connector110 in '/.../....svg' schematic of '/.../....fzp'
> ...connector111...
> ...connector112...
> ...connector113...
> ...connector114...
> ...connector115...
> ...connector116...
> ...connector117...
> ...connector100...
> ...connector101...
etc...

On Save:
> This part has 18 unassigned connectors in 1 view. Until all connectors are assigned to SVG elements, the part will not work correctly. Exiting the Parts Editor now is fine, as long as you remember to finish the assignments later.

ladyada, 2017-06-14:
hiya if you can submit a PR that would be rad - we've redone many but not all of our parts and we're a bit swamped so no ETA

andrewvaughan, 2017-06-14:
Unfortunately, I don't know enough about Fritzing to fix the issues stated.  I'll have to leave that to someone else.