adafruit/Fritzing-Library #103: Adafruit Perma-Proto HAT.fzpz is missing connections

RandScullard, 2017-07-23:
The "row of five" pads on the Adafruit Perma-Proto HAT.fzpz are not connected. If I connect a component to one pad in a row of five, and a wire to another in the same row, Fritzing does not show that they are hooked up.

kso512, 2017-07-23:
I believe I fixed this in my branch and pull request:

https://github.com/adafruit/Fritzing-Library/pull/101

Feel free to use the version from my Fritzing Library fork until the pull request is accepted:

https://github.com/kso512/Fritzing-Library/blob/master/parts/Adafruit%20Perma-Proto%20HAT.fzpz

RandScullard, 2017-07-23:
Thanks @kso512! That definitely fixes the problem.

paulcrawford, 2017-11-12:
@kso512, is there a way to further modify the HAT.fzpz to allow for trace cuts? To show a trace cut is fairly easy by putting a background colour line over the trace but that will not change the logic to show that the connection is actually broken. Presumably there is something in the .svg file that was changed to create the connection of the five pads. Can that be edited in the .svg file to create the trace cut?