adafruit/Fritzing-Library #114: Adafruit Particle/Spark NeoPixel Ring Kit missing

ELLIOTTCABLE, 2018-04-05:
The documentation for the Particle NeoPixel ring includes no information about the connections on the board (which leaves me with no information about which pins to solder, or whether there's 3.3v → 5v logic-level conversion for the Neopixels, or really anything else about how to use this? `:x`)

The [product page](https://www.adafruit.com/product/2268) claims there's Fritzing files in this project …

> * [EagleCAD PCB files available on GitHub](https://github.com/adafruit/Adafruit-Particle-Spark-Neopixel-Ring-PCB)
> * [Fritzing object in Adafruit Fritzing library](https://github.com/adafruit/Fritzing-Library)

… but I couldn't find it anywhere! (And I don't own EagleCAD …)

Please help?

ladyada, 2018-04-13:
its Adafruit NeoParticle Ring.fzpz

closed 2018-04-13 by ladyada

ELLIOTTCABLE, 2018-04-13:
Ahhhh; I was searching by ‘Spark’ instead of ‘Particle’, I suspect!

Your work is appreciated and you rock. 💖