adafruit/Fritzing-Library #124: RE: The eagle2fritzing Ideas and the Errors

silver2row, 2019-08-15:
Hello,

I am using Ubuntu and I am receiving many errors on the eagle2fritzing library ideas. 

...

I was wondering if there were any updates or ideas circulating on why my Ubuntu Desktop keeps showing that there is this, "", marking that keeps me from using make correctly. I am not sure if other people are having difficulty with this effort or not.

I just wanted to reach out and attempt to get some feedback.

Seth

P.S. I noticed there was no way to direct issues to that repo, i.e. I am here for that I guess. I will keep trying.

ladyada, 2019-08-15:
hiya, no idea - we barely support it, we just have the guide here
https://learn.adafruit.com/make-beautiful-fritzing-parts-with-eagle2fritzing-brd2svg

closed 2019-08-15 by ladyada

silver2row, 2019-08-15:
Hello,

Okay. No issue.

Seth