adafruit/Fritzing-Library #151: Feather RP2040 incorrect pin name

owenh000, 2021-10-03:
The Fritzing part at `parts/Adafruit Feather RP2040.fzpz` has the *VUSB* pin labeled as *VBUS* instead.

I can open a pull request if it will get merged (#101 for example hasn't been noticed since it was opened four years ago).

owenh000, 2021-10-03:
Actually I'll just drop the file here for anyone who may be interested. (`.zip` added so GitHub will accept it.)

[Adafruit Feather RP2040_fixed.fzpz.zip](https://github.com/adafruit/Fritzing-Library/files/7272877/Adafruit.Feather.RP2040_fixed.fzpz.zip)

dhalbert, 2021-10-03:
VBUS is the name on the schematic (https://learn.adafruit.com/assets/100337), and is common on our schematics for USB bus power. But the silk name is "USB", and our pinout diagram shows VUSB.

We would appreciate a PR -- thanks. We are working through old PR's, but there are a lot of them.

owenh000, 2021-10-03:
I opened pull request #152. Thanks! Also my comment about old pull requests seems a bit rude; sorry about that.

closed 2021-10-04 by kattni