adafruit/Fritzing-Library #154: LCD display component (ST7789)

command0r, 2022-01-04:
The guys here (https://www.hackster.io/wilderness-labs/build-your-own-temperature-monitor-with-meadow-edc696#) used Fritzing to produce schematics/diagram. I wonder where did they get a component for ST7789 (LCD display, IPS 240x240, 7pin)? It's not available in Fritzing by default and I can't find it anywhere to import (the actual .fzpz file). Please advice.

ladyada, 2022-01-04:
you should ask them :)

closed 2022-01-04 by ladyada

command0r, 2022-01-06:
@ladyada, I guess you closed this topic a bit prematurely.
It seems they just made up an image of this component, so it's not a real one adopted for Fritzing. I believe finger-pointing is not the best practice when you support the library of components to make this product usable. I paid for it and want proper support.
The question is - who can help to develop this component? I can see it's widely used with Arduino (see here: https://create.arduino.cc/projecthub/infoelectorials/project-020-arduino-1-3-ips-tft-display-project-d7a6ee), and not having it in the scheming tool is a big inconvenience.
I tried to figure out how the component designer works, but it seems it involves a lot more than putting some lines and shapes together. It requires some basic component blueprints developed in the vector graphic tools like Corel or Adobe Illustrator, which goes beyond my expertise. It looks like another potential area for improvement.

ladyada, 2022-01-15:
whats the exact product that you purchased from adafruit you're asking for support?