adafruit/Fritzing-Library #178: Announcement: DO NOT use E2F for schematic

RAPTOR7762, 2025-04-22:
A very, very, very important announcement. @BlitzCityDIY listen up:

## Please do not use the E2F tool. The schematic SVG produced is not correct.

The SVG produced lacks terminalIDs, so wires terminate to the centre of the pins. **It is undesirable**

## Use the Inkscape Schematic Extension instead!

ladyada, 2025-05-02:
we have to use eagle2fritzing - there is no other way for us to efficiently create devices. if you want you can submit a PR fix to eagle2fritzing for the change you want

RAPTOR7762, 2025-05-03:
I know, as it is easier :D I mean for the schematic view only, it is important that you use the [schematic extension](https://forum.fritzing.org/t/fritzing-schematic-an-inkscape-extension-released/13394), and not Eagle2Fritzing (really old, and there hasn't been an update in nearly a decade.)

RAPTOR7762, 2025-05-03:
OK @ladyada edited my post

closed 2025-05-24 by RAPTOR7762