adafruit/Fritzing-Library #181: PID:375 Magnetic contact switch (door sensor) Reed Switch

tyeth, 2025-05-01:
I've found a very similar MC38 product that I'm linking to as an alternative to ADA375, and it has a part here:
https://github.com/JZ-SmartThings/Fritzing/blob/master/Parts/MC-38.fzpz
Unfortunately there is no license so yanking it feels a bit wrong, but at least raising it as an issue here will hopefully allow others to find it more easily.

RAPTOR7762, 2025-05-02:
Please post this in the [Fritzing Forum](forum.fritzing.org)

tyeth, 2025-05-02:
Good shout.
Sadly realised it doesn't have pin connections so it's not the greatest part / model. I'm still using it in a learn guide, but will keep an eye out for a better one or maybe attempt to fix this one.

RAPTOR7762, 2025-05-02:
Unfory=tunately the above part is severely broken. Can you attatch the link to the Adafruit Website  of thia part?

RAPTOR7762, 2025-05-02:
I'll fix the part

RAPTOR7762, 2025-05-03:
Then maybe have to wait for @ladyada or @BlitzCityDIY to make a part (to respect the README) It's fine to fix their created parts tho